exelban/stats #1710: Log spamming about CacheDelete* when Disk sensor is enabled

luckman212, 2023-12-01:
**Details:**
 - Device: mac Mini 2020 M1
 - macOS: 14.1.2
 - Application version: 2.9.13

This is not a bug. But maybe there is a workaround or an explanation at least. In troubleshooting [another issue][1], I came across a large amount of log spew about Stats.app related to something about Disk Cache. 

Any idea what these things are? They relate somehow to querying the disk space when Disk sensor is enabled. My interval is 10s.

- `CacheDelete`
- `com.apple.cache_delete:client`
- `CacheDeleteCopyAvailableSpaceForVolume`
- `CacheDeleteVolume`

I searched the Stats sourcecode but couldn't really figure out what is causing it. Maybe `volumeAvailableCapacityForImportantUsageKey` calls? I found a [related discussion][2] that discusses the same issue with iStat Menus. I know the source code isn't the same but maybe the same API is being used causing this.

### Log output

Get Stats PID
```
$ pgrep -x Stats
1875
```

Get log stream from ULS
```
$ log stream --debug --info --predicate '(subsystem == "com.apple.cache_delete" && processID == 1875)' --style syslog
```

Here's 20 seconds of the output (it was too long to paste!) so here it is attached

[com.apple.cache_delete.txt](https://github.com/exelban/stats/files/13531824/com.apple.cache_delete.txt)

[1]: https://github.com/JohnCoates/Aerial/issues/1341#issuecomment-1836455839
[2]: https://eclecticlight.co/2022/09/08/what-shouldnt-you-see-in-your-xprocheck-results/#comment-81185

exelban, 2023-12-01:
Hi, thanks for the feedback.
It's not Stats themself. It's macOS. When calling the API that returns the free disk space. This part: 

https://github.com/exelban/stats/blob/96db37bb3b958298d223164df00d93462fd0b7a8/Modules/Disk/readers.swift#L115-L126

I spent some time to find another way to obtain a free space. But it's the only one that gives real size (that includes the purgeable space).

There is only 2 ways to omit these logs:
- disabling the Disk module (or increase the update interval, it will not prevent logs but make it less)
- showing the disk space without purgeable space -> but it will be incorrect data

luckman212, 2023-12-02:
Thanks @exelban 

I noticed max interval for Disk reader is 30sec. In my case I really would not mind having an interval of 1, 3, or 5 minutes even. It would help cut down on this log spam and my disk usage does not change much in a few minutes, more of a daily/weekly thing that I glance at.

closed 2023-12-10 by exelban

luckman212, 2023-12-10:
👍

Zorast, 2026-10-01:
I can reproduce a more severe variant of this behavior on Apple Silicon / macOS Tahoe.

Environment:
- MacBook Pro 16-inch, M1 Max
- 64 GB unified memory
- macOS Tahoe 26.x
- Stats with the Disk module enabled

In my case, this does not appear to be only harmless log spam.

After prolonged use with the Disk module enabled, the system can develop noticeable UI latency: scrolling with the built-in MacBook trackpad becomes jerky and multi-finger swipe animations between Spaces no longer feel smooth.

During troubleshooting, the behavior was isolated to the Stats Disk module. The relevant activity is consistent with what was originally reported in this issue:

```text
CacheDelete
com.apple.cache_delete:client
CacheDeleteCopyAvailableSpaceForVolume
CacheDeleteVolume
```

The Disk reader had been polling periodically, and disabling the Disk module entirely was the effective workaround. Increasing the polling interval reduces the frequency of the calls, but does not eliminate them.

What may be worth investigating is whether these repeated available-space queries can, under some conditions on Tahoe, cause secondary system activity or contention that goes beyond the Console log messages and eventually affects overall UI responsiveness.

This is particularly noticeable with:

- scrolling in multiple applications
- built-in trackpad scrolling
- multi-finger Space switching gestures
- general macOS UI animation smoothness

Restarting Dock or resetting the HID input path does not address the issue, which suggests that the physical trackpad/input layer itself is not the underlying problem.

Disabling the Stats Disk module is currently the reliable workaround.

I realize that the underlying `volumeAvailableCapacityForImportantUsageKey` / CacheDelete behavior comes from macOS, as explained earlier in this issue. However, given that on Tahoe this can apparently correlate with system-wide UI degradation rather than only log output, it may be worth reconsidering whether the purgeable-space query needs to run on every Disk refresh.

Possible mitigations could be:

- cache the purgeable-space result for a longer period than the normal Disk I/O refresh interval
- query purgeable/free capacity independently and much less frequently
- expose separate refresh intervals for disk I/O statistics and available-capacity calculations
- optionally disable the purgeable-space-aware query while keeping Disk I/O monitoring enabled

That would allow users to keep the Disk module active without continuously invoking the expensive macOS free-space path.