exelban/stats #1722: App Store Available?

haseeb-heaven, 2023-12-09:
Can you tell me that why this application is not available on the App Store for MacOS?
I couldn't find it on App Store if its available then, can you please share the link of this thanks.

exelban, 2023-12-10:
no, it's not available. First of all, it's a name: stats is already used in the App Store.
The second one it's quite difficult with the App Store app review. At the beginning of the project, I tried to deploy it to the App Store. But after a lot of rejects because the app does not fit some of the App Store rules I just gave up. Also, there are a lot of restrictions about updates, SMC (get sensors and set fan speed), etc. That from my perspective makes the version in the App Store just nonusable. 

PS: at some moment I saw seen fork of Stats that was available in the App Store. But it was the very first release that contained only a few features. After I started working on Stats more looked like it was impossible for the developers to make it work with app store rules. But it is just a guess.

closed 2023-12-10 by exelban

haseeb-heaven, 2023-12-10:
Yes i agree AppStore is very restrictive sometimes with these permissions.
I really like the app and with all its features outside appstore with restricted version