exelban/stats #1723: Wi-Fi network name (SSID) detection

CarterLi, 2023-12-11:
I saw that you removed WiFi network name from the network popup because of the OS limitation. You can actually fetch the information with [the private framework Apple80211](https://code.google.com/archive/p/iphone-wireless/wikis/Apple80211GetInfoCopy.wiki). If you don't want to use private frameworks, you may use `/usr/sbin/networksetup -getairportnetwork <inf_name>`. Both of them don't require location privileges.

exelban, 2023-12-11:
Could you give en example of how to get network name via `networksetup`?

exelban, 2023-12-11:
oh, I have check the wrong interface myself(
thx, will check what method is better and will add that.

CarterLi, 2023-12-11:
This is what I use: https://github.com/fastfetch-cli/fastfetch/blob/dev/src/detection/wifi/wifi_apple.m#L6-L25

The benefit of using `Apple80211` is that you don't need to spawn child processes. `networksetup` uses `Apple80211` internally. The only drawback is that it's private.

exelban, 2023-12-11:
yeap, it's a point. But also it's not check the interface details on each read (read interval). But only when the network is reconnected. I will check if there are benefits for Stats to use `Apple80211`.

closed 2023-12-24 by exelban

CarterLi, 2024-09-18:
Sadly, neither networksetup nor Apple80211 works in macOS Sequoia...