exelban/stats #1725: Blank app name rows in battery widget

andresrinivasan, 2023-12-14:
M2 Air on MacOS 14.1.1

[image]

exelban, 2023-12-14:
hi. Please share the output of `/usr/bin/top -o power -l 1 -n 15 -stats pid,command,power`.

andresrinivasan, 2023-12-14:
```sh
> /usr/bin/top -o power -l 1 -n 15 -stats pid,command,power
Processes: 434 total, 3 running, 431 sleeping, 4012 threads
2023/12/14 12:32:20
Load Avg: 2.63, 2.57, 2.27
CPU usage: 8.47% user, 13.61% sys, 77.91% idle
SharedLibs: 526M resident, 106M data, 38M linkedit.
MemRegions: 1108666 total, 4252M resident, 206M private, 1775M shared.
PhysMem: 15G used (2419M wired, 5908M compressor), 77M unused.
VM: 260T vsize, 4729M framework vsize, 10255527(0) swapins, 15063228(0) swapouts.
Networks: packets: 102191188/138G in, 38999206/48G out.
Disks: 116560786/1916G read, 78949860/3231G written.

PID    COMMAND          POWER
99956  icdd             0.0
99852  com.apple.WebKit 0.0
99846  Obsidian Helper  0.0
99844  Obsidian Helper  0.0
99843  Obsidian Helper  0.0
99842  com.apple.WebKit 0.0
99840  com.apple.WebKit 0.0
99839  com.apple.WebKit 0.0
99837  com.apple.WebKit 0.0
99836  com.apple.WebKit 0.0
99831  commerce         0.0
99821  diagnostics_agen 0.0
99816  avconferenced    0.0
99813  mediaremoteagent 0.0
99806  PAH_Extension    0.0
```

exelban, 2023-12-15:
thx. Nope, one read is enough. Because it's a problem with parsing the output line.

closed 2023-12-17 by exelban