exelban/stats #1727: Some process names not showing up in Battery section

XInTheDark, 2023-12-16:
**Describe the bug**
In the Battery panel, the individual names of running processes are no longer shown. As seen below:
[image]

On the other hand, the names are shown correctly in other panels, e.g. CPU info. As shown:
[image]

Additionally I don't know the exact version where the bug started appearing, but this behaviour only occurs in recent versions and before an automatic update, it was working correctly. If there is a way to check perhaps some log to see which was my previous version, I'll be happy to assist.

I have tried restarting Stats and the Mac but both do not resolve the issue.

**Details:**
 - Device: MacBook Air M1, 2020
 - macOS: Sonoma 14.1
 - Application version: 2.9.15 (latest)

exelban, 2023-12-16:
rly?

closed 2023-12-16 by exelban

F1248, 2023-12-16:
I can confirm the problem and don't understand why it was closed as not planned:
MacBook Air M1 (`MacBookAir10,1`)
macOS Sonoma 14.2 (`23C64`)
Stats Version 2.9.15