exelban/stats #1731: iGPU monitor shows 0% utlisation

Datisit, 2023-12-21:
**Describe the bug**

I can't seem to get the iGPU monitor to display any utilisation.

At first I though it might be cause I didn't have Power Gadget installed, and instead had PCM on my system. 
But now that I've uninstalled PCM (altough I can't seem to get rid of the kext), and re-installed the Intel Power Gadget, there's still no utilisation displayed in the Stats. 
[image]

Utilisation is being displayed correctly within the Power Gadget window:
[image]

**Details:**
 - Device: [e.g. Hackintosh SMBIOS: iMac 2019 5K]
CPU: Intel 8700k
GPU: Sapphire Nitro+ RX 580 8GB SE
RAM: 64 GB (4x16 3600) G-Skill F4-3600C16
Motherboard: ASUS Prime z370-a II
 - macOS: [e.g. 12.7.2]
 - Application version: [e.g. 2.9.16]

exelban, 2023-12-23:
Stats just show the value that macOS has. In this case probably the driver does not report the value to the system.

closed 2023-12-23 by exelban

Datisit, 2023-12-23:
Ah, ok. Do you know which driver would be responsible?

And would there be anyway to get the value in the same way that the Intel power gadget does?