exelban/stats #1733: How to interpret internet connection "DOWN"?

tulpixer, 2023-12-21:
I'm having trouble understanding the entry for internet connection (in German Internetverbindung). What could be the cause for me getting red dots (meaning DOWN) every few seconds? 
[image]

Chiroyce1, 2023-12-21:
You aren't receiving proper responses for the pings sent to the connectivity host, the red squares indicate packet loss.

tulpixer, 2023-12-22:
Is it possible that this connectivity host is down, or can I be certain that the problem is on my side?

exelban, 2023-12-22:
probably there is a problem somewhere on your side if you do not change the ICPM host

Chiroyce1, 2023-12-22:
@tulpixer Try out different hosts and see if the problem persists, try them out by simply using `ping` in the terminal. (`-c` is the number of pings and `-i` is the interval between them)
```bash
ping -c 20 -i 0.5 1.1.1.1   # cloudflare
ping -c 20 -i 0.5 8.8.8.8   # google dns
# ... and so on with other IPs or hosts you can find
```

tulpixer, 2023-12-22:
Aah, okay, now I understand that the ICMP host in Stats' settings is IP address for the ping test. As 1.1.1.1 is Cloudflare, a downtime on their side is obviously not very probable. Thank you very much!

closed 2023-12-23 by exelban