exelban/stats #1741: Feature Request: Allow hiding fans in the sensors popup entirely

Kiiyya, 2023-12-25:
I love this app, thank you for making it!

It is possible to disable all options for sensors *except* the fans, where you can only choose between "RPM" and "Percentage".
[image]

My feature request is simple: Add a "None" option to hide the fans from the popup.

I only use this app to display my total system power consumption.

exelban, 2023-12-27:
you can achieve that by disabling all sensors.

Kiiyya, 2023-12-27:
If I disable the sensors entirely, I won't be able to view the power consumption anymore, which is the reason why I use this app. So this does not solve my issue.

Although I understand if this is very far down your priority list, and only a minor annoyance I can live with. You still did a great job :).

exelban, 2023-12-27:
oh, I misunderstand the problem. I was thinking about disabling sensors, not fans. Yeap, I can add an option to hide the fans in the same way as sensors.

closed 2023-12-31 by exelban

Kiiyya, 2024-01-01:
Thank you so much, this is perfect <3.