exelban/stats #1742: [Feature Request] Options to de-clutter and customize the network pop-up

victor-marino, 2023-12-26:
Let me preface this by saying that I love your work. The widgets are extremely customizable and better than the paid alternatives in many ways.

I'm personally very interested in the 'network' widget, but I find the pop-up to be a bit cluttered in its current form. Some ideas for improvement:
- Allow us to choose which elements should be part of the pop-up widget, and maybe rearrange them too. For instance, if I'm already seeing the network speed numbers in the status bar itself, I don't really need to see it again in large numbers in the pop-up. Likewise, some people may prefer to disable the panels for connectivity history, details, processes, etc. Some people may also want to show the public IP address at the top, etc.
- For the active panels, I think some of the info could also be hidden behind hover menus. E.g.: we could remove the WiFi-specific stuff such as the MAC address, SSID, signal strength, Wi-Fi standard, frequency channel, etc., and only show those things when we hover over the Wi-Fi interface, as a new pop-up.

I think these little tweaks would help clean up the widget and show the key information at a glance, while keeping the details easily available whenever you need them.

In addition to all this, I see there's already [this issue](https://github.com/exelban/stats/issues/1194) to show a larger chart with the traffic history when hovering as well, which I wholeheartedly support as well!

hastebrot, 2023-12-29:
yes, hiding the MAC address ("physical address") in the network pop-up or public IP is also useful, when you share your screen and accidentally click on the network pop-up.

exelban, 2024-01-08:
Hi. Thanks for your feedback.
I don't see it's like a good feature. Stats already has too many options that are possible to configure. I don't see any reason to make the popup view configurable as well. There is enough space to fit everything possible to show.

If there is too much information you can easily make a build with only interesting information for you.

closed 2024-01-08 by exelban