exelban/stats #1748: Combined modules incomplete

ibridgepeople, 2023-12-28:
Hi, 
I'm using your fantastic modules. 
On the combined view, I have only five: CPU, GPU, RAM, Disk, and Network. 
I tried to add Sensors and a Clock.
I may have missed something in the instructions. 
Thanks for your help.
Kind regards,
Marcelo
marcelo.brida@gmail.com

1. MacBook Pro
2. 16", 2019
3. 2,3 GHz Intel Core i9 8 cœurs
4. AMD Radeon Pro 5500M 8 Go
5. Intel UHD Graphics 630 1536 Mo
6. 64 Go 2667 MHz DDR4
7. DD 2 To
8. mac OS 14.2.1 (23C71)

[image]

exelban, 2023-12-28:
Hi. Yeap, there are no views for sensors and clock modules in the combined view.
The clock could be added. But I'm not sure I will add a sensor view. It's something that will be done when I will redesign the combined view.

ibridgepeople, 2023-12-28:
Ok, thanks a lot

closed 2023-12-31 by exelban