exelban/stats #1759: Stats crashes frequently

Traace, 2024-01-03:
**Describe the bug**
1. Run stats for a few minutes
2. Get a crash with a MacOS error message

I run into this issue every time I'm using stats.

**Details:**
 - Device: Mac Studio M1 Ultra
 - macOS: 14.2.1
 - Application version: 2.9.18
 - 3 Network Devices: BuiltIn, WiFi, RJ-45 GB-Ethernet USB
 - 4 Storage Devices: BuiltIn, 2x USB, SD-Card
 - 6 External Storage Devices: 2x SMB, 4x SSHFS

**Crash Log:**

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               Stats [782]
Path:                  /Applications/Stats.app/Contents/MacOS/Stats
Identifier:            eu.exelban.Stats
Version:               2.9.18 (518)
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2024-01-03 11:38:14.2691 +0100
OS Version:            macOS 14.2.1 (23C71)
Report Version:        12
Anonymous UUID:        A71705C4-EAD5-0C8A-675C-51BA8008B00B

Time Awake Since Boot: 990 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BREAKPOINT (SIGTRAP)
Exception Codes:       0x0000000000000001, 0x00000001999216c8

Termination Reason:    Namespace SIGNAL, Code 5 Trace/BPT trap: 5
Terminating Process:   exc handler [782]

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libswiftCore.dylib            	       0x1999216c8 swift_unknownObjectRetain + 44
1   libswiftCore.dylib            	       0x19960e3ac String._bridgeToObjectiveCImpl() + 96
2   Kit                           	       0x104d42854 0x104ca8000 + 632916
3   Kit                           	       0x104d41ab4 0x104ca8000 + 629428
4   Kit                           	       0x104d41d40 0x104ca8000 + 630080
5   AppKit                        	       0x18d65bc60 _NSViewDrawRect + 124
6   AppKit                        	       0x18dfd2c8c -[NSView _recursive:displayRectIgnoringOpacity:inContext:stopAtLayerBackedViews:] + 1088
7   AppKit                        	       0x18db74c60 __49-[NSViewBackingLayer _renderForegroundInContext:]_block_invoke + 120
8   AppKit                        	       0x18d65b4d4 -[NSView(NSLayerKitGlue) _drawViewBackingLayer:inContext:drawingHandler:] + 556
9   AppKit                        	       0x18db74b94 -[NSViewBackingLayer _renderForegroundInContext:] + 96
10  QuartzCore                    	       0x191e9b2a8 -[CALayer renderInContext:] + 1200
11  QuartzCore                    	       0x191e9cb18 -[CALayer _renderSublayersInContext:] + 340
12  QuartzCore                    	       0x191e9b2b4 -[CALayer renderInContext:] + 1212
13  QuartzCore                    	       0x191e9cb18 -[CALayer _renderSublayersInContext:] + 340
14  QuartzCore                    	       0x191e9b2b4 -[CALayer renderInContext:] + 1212
15  QuartzCore                    	       0x191e9cb18 -[CALayer _renderSublayersInContext:] + 340
16  QuartzCore                    	       0x191e9b2b4 -[CALayer renderInContext:] + 1212
17  QuartzCore                    	       0x191e9cb18 -[CALayer _renderSublayersInContext:] + 340
18  QuartzCore                    	       0x191e9b2b4 -[CALayer renderInContext:] + 1212
19  AppKit                        	       0x18dfd238c -[NSView _renderLayerInContext:] + 204
20  AppKit                        	       0x18dfd2a58 -[NSView _recursive:displayRectIgnoringOpacity:inContext:stopAtLayerBackedViews:] + 524
21  AppKit                        	       0x18dfccd4c __47-[NSView displayRectIgnoringOpacity:inContext:]_block_invoke + 656
22  AppKit                        	       0x18d584714 NSPerformVisuallyAtomicChange + 108
[... 565 lines cut ...]
USB Device: USB31Bus
USB Device: JMS579
USB Device: USB31Bus
USB Device: USB31Bus
USB Device: USB C Video Adaptor
USB Device: USB31Bus
USB Device: Generic Billboard Device
USB Device: USB31Bus
USB Device: USB2.0 Hub
USB Device: USB2.0 Hub
USB Device: USB 2.0 Hub
USB Device: Jabra BIZ 2400 II CC
USB Device: Wireless Mouse
USB Device: Lenovo Traditional USB Keyboard
USB Device: Stream Deck XL
USB Device: USB3.0 Hub
USB Device: USB 10/100/1000 LAN
USB Device: USB3.1 Hub
Thunderbolt Bus: Mac Studio, Apple Inc.
Thunderbolt Bus: Mac Studio, Apple Inc.
Thunderbolt Bus: Mac Studio, Apple Inc.
Thunderbolt Bus: Mac Studio, Apple Inc.
Thunderbolt Bus: Mac Studio, Apple Inc.
Thunderbolt Bus: Mac Studio, Apple Inc.
```

exelban, 2024-01-03:
Hi. Please disable this (screenshot) widget in all modules.
[image]

Traace, 2024-01-03:
Hi, thanks for your help. I did disable this widget in all modules. However the crash still occures, but this time exception type is EXC_BAD_ACCESS (SIGSEGV)

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               Stats [15349]
Path:                  /Applications/Stats.app/Contents/MacOS/Stats
Identifier:            eu.exelban.Stats
Version:               2.9.18 (518)
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2024-01-03 14:13:50.3593 +0100
OS Version:            macOS 14.2.1 (23C71)
Report Version:        12
Anonymous UUID:        A71705C4-EAD5-0C8A-675C-51BA8008B00B

Time Awake Since Boot: 10000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x000035b27edb06e0
Exception Codes:       0x0000000000000001, 0x000035b27edb06e0

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [15349]

VM Region Info: 0x35b27edb06e0 is not in any region.  Bytes after previous region: 58559712397025  Bytes before following region: 46512367532320
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      commpage (reserved)        1000000000-7000000000   [384.0G] ---/--- SM=NUL  ...(unallocated)
--->  GAP OF 0x5f9000000000 BYTES
      MALLOC_NANO              600000000000-600020000000 [512.0M] rw-/rwx SM=PRV  

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libswiftCore.dylib            	       0x1999216cc swift_unknownObjectRetain + 48
1   libswiftCore.dylib            	       0x19960e3ac String._bridgeToObjectiveCImpl() + 96
2   Kit                           	       0x102e52854 0x102db8000 + 632916
3   Kit                           	       0x102e51ab4 0x102db8000 + 629428
4   Kit                           	       0x102e51d40 0x102db8000 + 630080
5   AppKit                        	       0x18d65bc60 _NSViewDrawRect + 124
6   AppKit                        	       0x18dfd2c8c -[NSView _recursive:displayRectIgnoringOpacity:inContext:stopAtLayerBackedViews:] + 1088
7   AppKit                        	       0x18d65b4d4 -[NSView(NSLayerKitGlue) _drawViewBackingLayer:inContext:drawingHandler:] + 556
8   AppKit                        	       0x18db74afc -[NSViewBackingLayer drawInContext:] + 56
9   AppKit                        	       0x18dd55ce0 -[NSViewBackingStore update:] + 560
10  AppKit                        	       0x18db73cf8 -[NSViewBackingLayerContents update:] + 52
11  AppKit                        	       0x18db74730 -[NSViewBackingLayer display] + 1240
12  QuartzCore                    	       0x191e9ae84 -[CALayer renderInContext:] + 140
13  QuartzCore                    	       0x191e9cb18 -[CALayer _renderSublayersInContext:] + 340
14  QuartzCore                    	       0x191e9b2b4 -[CALayer renderInContext:] + 1212
15  QuartzCore                    	       0x191e9cb18 -[CALayer _renderSublayersInContext:] + 340
16  QuartzCore                    	       0x191e9b2b4 -[CALayer renderInContext:] + 1212
17  QuartzCore                    	       0x191e9cb18 -[CALayer _renderSublayersInContext:] + 340
18  QuartzCore                    	       0x191e9b2b4 -[CALayer renderInContext:] + 1212
19  QuartzCore                    	       0x191e9cb18 -[CALayer _renderSublayersInContext:] + 340
20  QuartzCore                    	       0x191e9b2b4 -[CALayer renderInContext:] + 1212
21  AppKit                        	       0x18dfd238c -[NSView _renderLayerInContext:] + 204
22  AppKit                        	       0x18dfd2a58 -[NSView _recursive:displayRectIgnoringOpacity:inContext:stopAtLayerBackedViews:] + 524
23  AppKit                        	       0x18dfccd4c __47-[NSView displayRectIgnoringOpacity:inContext:]_block_invoke + 656
24  AppKit                        	       0x18d584714 NSPerformVisuallyAtomicChange + 108
25  AppKit                        	       0x18d78b8b4 -[NSView displayRectIgnoringOpacity:inContext:] + 104
26  AppKit                        	       0x18d78b7fc __62-[NSView cacheDisplayInRect:toBitmapImageRep:includeSubviews:]_block_invoke + 192
27  AppKit                        	       0x18d78b718 -[NSBitmapImageRep _captureDrawing:] + 72
28  AppKit                        	       0x18d78b6b4 -[NSView cacheDisplayInRect:toBitmapImageRep:includeSubviews:] + 168
29  AppKit                        	       0x18dd152a0 -[NSStatusItem _updateReplicant:] + 1168
[... 523 lines cut ...]
USB Device: USB31Bus
USB Device: JMS579
USB Device: USB31Bus
USB Device: USB31Bus
USB Device: USB C Video Adaptor
USB Device: USB31Bus
USB Device: Generic Billboard Device
USB Device: USB31Bus
USB Device: USB2.0 Hub
USB Device: USB2.0 Hub
USB Device: USB 2.0 Hub
USB Device: Jabra BIZ 2400 II CC
USB Device: Wireless Mouse
USB Device: Lenovo Traditional USB Keyboard
USB Device: Stream Deck XL
USB Device: USB3.0 Hub
USB Device: USB 10/100/1000 LAN
USB Device: USB3.1 Hub
Thunderbolt Bus: Mac Studio, Apple Inc.
Thunderbolt Bus: Mac Studio, Apple Inc.
Thunderbolt Bus: Mac Studio, Apple Inc.
Thunderbolt Bus: Mac Studio, Apple Inc.
Thunderbolt Bus: Mac Studio, Apple Inc.
Thunderbolt Bus: Mac Studio, Apple Inc.
```

exelban, 2024-01-03:
hmm, do you have a combined module enabled?

exelban, 2024-01-03:
also, there is an option to paste a file. Please don't paste so large text. It's impossible to read anyway.

Traace, 2024-01-04:
Yes, combined module was enabled. I disabled it and there are no more crashes.

Thank you. Closed.

closed 2024-01-04 by Traace

exelban, 2024-01-04:
Do you want to help to fix that maybe?
Because the bug exists, but I cannot debug it locally(