exelban/stats #1764: Constant 10% CPU usage, is that normal?

tbergeron, 2024-01-08:
**Describe the bug**
Maybe it's normal for Stats but I'm an iStat Menus user and that app uses around 1-2% CPU so I thought 10% was quite high. Is that normal?

Here's an example screenshot to see how much CPU is being used at all times. It lowers at around 8% but never goes lower even when there are no Stats window opened at all.

Any tips on how to lower it? I already set the update interval to 3 secs for CPU and RAM. 

I'm so glad there is finally an alternative and I'm willing to help as much as I can to improve Stats (I'm a software dev myself) so keep up the great works y'all!

Thanks!

[image]

**Details:**
 - Device: MacBook Air M1
 - macOS: macOS Ventura (13.6.3)
 - Application version: Version 2.9.19

exelban, 2024-01-08:
It depends on how many modules you have and what intervals are set up.
Also if you have an opened popup Stats takes some CPU to draw everything. 
For closed popups, it's normal around 6% CPU usage (activity monitor) for the Apple silicon since it's mostly on e-cores.

iStats splited into multiple processes so the main process is quite light because it's only for visualization. Stats it's one process only.

TL.DR. Yes if you have opened the popup.

tbergeron, 2024-01-08:
> It depends on how many modules you have and what intervals are set up. Also if you have an opened popup Stats takes some CPU to draw everything. For closed popups, it's normal around 6% CPU usage [...]

CPU usage is indeed a bit lower when no Stats window / menu are opened but it still has ~3 times more CPU usage than iStat Menus. As seen on these screenshots taken from Activity Monitor:

[image]

vs

[image]

This is my CPU widget configuration (same intervals as I have in iStat Menus)

[image]

Any suggestions? I'm just trying to minimize the CPU usage to minimize battery usage, etc.

Thanks for your quick reply!

exelban, 2024-01-08:
with all modules enabled it takes around 6% CPU, if you want less usage just disable some modules. More info you can find in the readme.