exelban/stats #1767: Apple M3 Pro 11C Temperature Sensors

xdsta, 2024-01-12:
Hey, I got my new M3 Pro 11C Macbook and it seems, that Stats could not identify every temperature sensor. How can I help you to gather the needed information? Which logs/exports can I provide? :)

[image]

**Details:**
 - Device: Macbook Pro M3 pro 11C
 - macOS: 14.2.1 (23C71)
 - Application version: Version 2.9.19

xdsta, 2024-01-12:
Just found a similar request and exported this file for you in advance :)

[m3pro11c.txt](https://github.com/exelban/stats/files/13916458/m3pro11c.txt)

djcristi, 2024-01-13:
some info also here: [issue 1703](https://github.com/exelban/stats/issues/1703)

closed 2024-01-14 by exelban

xdsta, 2024-01-14:
Just to understand correctly. It is not planned to support the newest generation of Apple Silicon? I am a little confused.

exelban, 2024-01-14:
there is an opened issue related to that.

xdsta, 2024-01-14:
Oh okey, I thought https://github.com/exelban/stats/issues/1703 was already closed so I did not comment there. Sorry.