exelban/stats #1772: m3 SoC sensors

exelban, 2024-01-14:
Hi.
It's an M3 SoC sensors discussion.
Looks like Apple started to have unique sensor keys for each version of SoC I can't support a sensor list for all SoCs. So if you want to help with debugging the sensor keys you can do it here without opening a new issue.

I cannot help with getting sensor keys since I don't have each platform. And sending me a list of keys does not help either. The best way so far was described by @tycho (#1703).

Please do not open a new issue for m3/m3 pro/max/ultra. And ping me when you will have new sensors that could be added to the Stats (or you can do it yourself with PR).

xdsta, 2024-01-14:
M3 Pro 11C [m3pro11c.txt](https://github.com/exelban/stats/files/13931837/m3pro11c.txt)

xdsta, 2024-01-14:
I exported all my values and checked if the number of sensors with similar names matches my core-count.

So my guess would be:

E-Cores (6)
```
[Tp3O]     45.93525695800781
[Tp3P]     48.796875
[Tp3S]     47.10912322998047
[Tp3T]     47.375
[Tp3W]     45.47585678100586
[Tp3X]     50.5625
```

P-Cores (5)
```
[TVS0]     32.56657791137695
[TVS1]     31.921131134033203
[TVS2]     31.370746612548828
[TVS3]     31.346330642700195
[TVSx]     32.56657791137695
```

I could not find a "group" of sensors that matches the GPU-Cores (14). If I divide them by uppercase/lowercase then I would guess this group:

```
[Tp0a]     41.93525695800781
[Tp0b]     49.23525619506836
[Tp0c]     45.640625
[Tp0g]     41.774715423583984
[Tp0h]     49.07471466064453
[Tp0i]     48.796875
[Tp0m]     41.51588439941406
[Tp0n]     48.81588363647461
[Tp0o]     44.5625
[Tp0u]     42.86343765258789
[Tp0v]     52.16343688964844
[Tp0w]     47.375
[Tp0y]     43.10900115966797
[Tp0z]     52.409000396728516
```

Tp0% also includes this 15 sensors, but they are with numbers/uppercase instead:

```
[Tp04]     40.56461715698242
[Tp05]     46.664615631103516
[Tp06]     45.046875
[Tp0C]     40.009849548339844
[Tp0D]     46.10984802246094
[Tp0E]     44.6875
[Tp0K]     40.18094253540039
[Tp0L]     46.280941009521484
[Tp0M]     46.234375
[Tp0R]     40.17270278930664
[Tp0S]     46.272701263427734
[Tp0T]     44.390625
[Tp0U]     41.59812545776367
[Tp0V]     48.89812469482422
[Tp0W]     44.859375
```

Anyone any experience with the naming scheme? Thank you.

xdsta, 2024-01-14:
[image]

looks like TG Pro could handle the new CPUs. Tried to match the temperatures with the unknown sensors but could not find a logical match :(

IT-Kwj, 2024-01-29:
The m3 max version of the network will show that it can be used normally when VPN is turned on, but the download speed will not be displayed when VPN is turned off

djcristi, 2024-01-31:
@exelban are you planning to fix the m3 issues i've described here? https://github.com/exelban/stats/issues/1703#issuecomment-1890922361

exelban, 2024-01-31:
@djcristi do you read the issue?

djcristi, 2024-01-31:
yes, i think me and the others already provided some infos/details on the other issue/link. for m3 max 14" 14c/ 16" 14c  at least i think i've provided.

exelban, 2024-01-31:
Looks like I have missed the mapping between the sensor keys and the names you are talking about. Please share it again.

Gea97, 2024-02-03:
Hi, I want to help with this but I don't know how

I also have TG Pro, is there a way to extract the list from there?

Foul-Tarnished, 2024-02-15:
I have a M3 Max 16"
the 30C gpu variant

if you don't have it already

closed 2024-08-12 by exelban