exelban/stats #1784: Bug: Only upload speed shows on bar

yukkodesu, 2024-01-18:
Like picture showed below, only upload speed is calculated at network activity. 

In the network activity details of each process, more than one process exhibits download speed, but the overall download speed is always zero.

I'm using a m3 pro macbook pro on sonoma 14.2.1, stats version 2.9.20. Restarting macOS and reinstalling did not solve it.

[image]

[image]

xiaozealer, 2024-01-19:
are you using vpn? It happens to me when I'm connected to a VPN

80211WiGuy, 2024-01-19:
Came here with the same issue - and yes I am using VPN (FortiClient).  This also seems to be an issue with the built-in System Monitor - Network tab so the issue is likely at the OS level with Sonoma (I'm on 14.2.1 (23C71) currently).

The issue seems present when not connected to VPN also, but...... Heyyyy, I found it seems to be related to the "Content Filter" installed by my forticlient that shows up in Network-> VPN & Filters.  When I disable it for the 5 seconds or so it allows, the downstream speed starts reporting normally until the content filter gets turned back on automatically.  I assume the forticlient app is doing this.

yukkodesu, 2024-01-20:
it works normally after disabled filters in Network-> VPN & Filters

closed 2024-01-21 by exelban