exelban/stats #1786: Incorrect temperature alert notification

valynor, 2024-01-20:
Mac Studio M1 Max, macOS 14.2.1, Stats 2.9.20
Sensors > Notifications has two alerts set: 
- Hottest CPU: 80°C
- Hottest GPU: 80°C

While watching a movie on Disney+ I suddenly got a Stats notification with a temperature alert, it disappeared so quickly I didn't have time to read it completely - but it was either the CPU or GPU 80°C alert. 

When I looked into the notification center the alert was gone, even though I have set Stats notifications as Alerts in system settings: they should stay on screen until I dismiss them and they certainly should not disappear from the notification center by themselves. 

The mac was doing nothing except showing the video in the browser, all temperatures were nominal around 40°C when I looked at them a couple seconds after the alert. No fans audible, nothing unusual. 

I'm pretty sure there's a bug in Stats regarding these notifications, the Mac Studio has very strong cooling and the highest temperature I've ever seen yet was 58°C while running a benchmark and that was in the high summer with a room temperature of over 30°C. 

This disappearing notification alert happened once before a few weeks ago.

exelban, 2024-01-23:
When the temperature goes below the threshold the notification is deleted.
Did you even try to see the history of this sensor? At least to ensure that the temperature was below the threshold when you received the notification.

valynor, 2024-01-23:
I have not yet done that, frankly I forgot that you can click the sensors to get the graph. I will check it next time it happens. 

But I don't think it's possible for the GPU to jump to 90°C for only a second with basically no load on it. I have the sensor alerts active for several weeks and this only started a few days ago, around stats version .19 or .20

I've also run https://browserbench.org/MotionMark1.3/ in the same browser I watch Disney+ in with the sensors popup open and all GPU cores stayed below 45°C at all times *while running a GPU benchmark*.

valynor, 2024-02-01:
@exelban 
I don't have a screenshot yet of a overheating alert but I just had this artefact in the "hottest GPU" graph, again while watching a video on D+: 

[image]

Any idea what could be causing this?

exelban, 2024-02-01:
hmm, and do you maybe notice what value it was in the notification?
or what threshold do you have for sensor notifications?

exelban, 2024-02-01:
please share the hottest GPU notification threshold

valynor, 2024-02-01:
There was no notification this time, I just accidentally saw this "negative" spike. 

The 2 notifications I have set are: 

[image]

exelban, 2024-02-01:
what do you mean by `There was no notification this time`?

valynor, 2024-02-01:
When I first reported this I had a temp alert notification for the GPU being supposedly at >90°C - which I frankly think is impossible in a Mac Studio during normal use. I can't even get the GPU temp >60° if I deliberately try to by running taxing benchmarks. 

This time there was a negative spike in the graph that's also not really possible, the GPUs can't drop from ~37°C to <10°C for just one second. 

Either I have some issue with my hardware temp sensors or Stats has a bug reading out the sensor and producing both too high and too low values occasionally.

exelban, 2024-02-01:
Stats do only one with sensors: read and show. That's all. If you have strange sensor values it has nothing to the Stats.

valynor, 2024-02-01:
"My software can't be erroneous" is the funniest thing I heard in a long time. 
There are dozens of things that could go wrong reading out and processing data.