exelban/stats #1787: Add option to set a fixed max value for the scale of the network charts

victor-marino, 2024-01-21:
In the network charts, it would be great to have the option of fixing the scale manually, as many other apps allow us to do.

I've created https://github.com/exelban/stats/pull/1746 as an initial implementation. 

Among other things, this would allow us to use the network chart as a "utilization" chart, where the chart represents the % of bandwidth we're using out of the total available bandwidth in our internet connection.

A more detailed explanation of why this feature is useful can be found [here](https://github.com/exelban/stats/pull/1746#issuecomment-1890782445).

The core logic is done with this PR and the chart plots the values as expected. However, there are some missing tweaks that I couldn't implement by myself due to lack of knowledge about MacOS development.

What's already implemented:
* New toggle to enable/disable the Custom Scale option
* New text field to enter the maximum desired bandwidth for the chart in Mbps (for now, a common value for DL and UL)
* Independent settings have been added for the widget and the pop-up charts
* Both charts will pick up the new value as soon as the text field is modified (if the toggle is enabled)

Still to do, hopefully with @exelban's help 😉 :
* Hide/disable the input text field when the toggle is disabled. Of course, up to @exelban to decide if he prefers an alternative behaviour. For instance, we could instead remove the toggle, and then:
  * Enable the feature if the user enters a bandwidth value
  * Disable the feature if the user leaves the text field empty
* Sanitize the input of the text field to allow only numerical digits
* Allow entering separate values for the download and upload speed

closed 2024-05-05 by exelban

victor-marino, 2024-05-06:
Thanks for this! Working great :)

What about the status bar widget? Any chance the scale can be fixed in there as well?

exelban, 2024-05-06:
yeap, it will be added

knokknok, 2026-10-28:
Has this been added to the widget? I can't seem to find the option.