exelban/stats #1788: Tooltips in network chart are cut off

zigapovhe, 2024-01-21:
**Describe the bug**
Tooltip in network chart is cut off. Also, i'm not sure, if it is just cut off like that, but sometimes it seems that tooltip shows "Kt".

**Details:**
 - Device: Macbook Pro 2021 M1 Pro 14'
 - macOS: 14.2.1
 - Application version: v2.9.21

[image]
[image]
[image]

zigapovhe, 2024-01-21:
Same issue if I switch to Byte as main unit. Tooltip gets randomly cut off.

exelban, 2024-01-22:
it's not randomly cutting off. The tooltip has a static width that fits only a few symbols.

zigapovhe, 2024-01-28:
Confirming bug was resolved in the latest update. Thank you. Closing.

closed 2024-01-28 by zigapovhe