exelban/stats #1789: Some Widgets occasionally missing on bootup

valynor, 2024-01-22:
Mac Studio M1 Max, macOS 14.2.1, Stats 2.9.20

I just booted my Mac and when Stats started up the CPU widget was missing in the menu bar. All of the other widgets (sensors, GPU, RAM, Drives, Network) where present and working. 
Quitting and restarting Stats fixed it. 

I had this happen before about 1 month ago, at that time only the Network widget was missing.

xeron, 2024-01-22:
Same, CPU widget often disappears for me too on M2 mac. Toggling the widget off/on fixes it for me.

FabulousCupcake, 2024-01-23:
I've been having the same issue since several weeks ago as well

* I thought it was not playing nice with [Bartender](https://www.macbartender.com/Bartender5/), but finally got around to test a bit and even with it disabled, some widgets are usually missing (sometimes all)
* There's no particular buggy widget — sometimes all of them are gone (I have cpu, disk, and sensor)
* Quitting and starting Stats.app again do not always fix it
* Going to configuration page and hitting ⏸️ and then ▶️ always fixes it

exelban, 2024-02-11:
hi. With the last update that problem must disappear.

closed 2024-02-11 by exelban