exelban/stats #1794: tempeture indicator is freezing

ozgurtaylankarakoc, 2024-01-25:
after the update 2.9.18 , tempeture indicator on the menubar is freezing, like never before. it was getting update like dynamically  but now even i rebooted the app or pc or both , indicator on the menubar is just laying wrong and static 
[image]

exelban, 2024-01-25:
please share historical data (chart) for any random sensors

ozgurtaylankarakoc, 2024-01-25:
[image]

charts re working without issue

exelban, 2024-01-25:
oh, sorry missed the problem with the menu bar. Will check that.

exelban, 2024-01-25:
hmm, what sensor you have active?

ozgurtaylankarakoc, 2024-01-25:
i didnt specify any sensor for it  it was standing as a default settings, before that updates Menubar indicator was following the hottest cpu right ? as i followed it was like that. so long story is short last several updates,
 its not following hottest cpu dynamically but as i observed its showing the just peak of the cpu tempeture after app launched.

https://github.com/exelban/stats/assets/117117615/ea2a8e7f-03f4-4c31-bd25-9668a1811729

exelban, 2024-01-25:
ok, so as you can see it's not a problem with the widget. You don't have any widgets enabled. Looks like you have some sensors enabled that are not available now. And it appears only on start and does not receive an update after, because there are no sensors like that.

I need to clean up the unavailabe sensors from list of enabled on app start.

ozgurtaylankarakoc, 2024-01-25:
thanks for responding and dealing with this issue, it was the small but annoying. 🫡

exelban, 2024-02-19:
Hi.
I have tried to fix that. But I figured out that is not a problem. Basically, you need to hide those "not updating" sensors from the list. But since it's not visible on the list you need to enable the "Unknown sensors" option, find the active sensors, and disable it. It will remove that sensor from the widget.

closed 2024-02-19 by exelban

ozgurtaylankarakoc, 2024-02-21:
I realized also when you warned me about unknown sensors then i checked pop-up menu which ones are available when turned on at the modul menu. it has began the appear dynamically again as like before, thanks for your support