exelban/stats #1795: Support more sensors

liangqi, 2024-01-26:
**Describe the bug**

There are too many unknown sensors in Stats, comparing with TG Pro, could we support them? Thanks.

**Details:**
 - Device: MacBook Pro 2019 16", Intel
 - macOS: 14.2.1 (23C71)
 - Application version: 2.9.21

[image]

closed 2024-01-26 by exelban

liangqi, 2024-01-26:
If no plan from devs, any guide or hint on how to improve it if someone else wants to do? Thanks.

liangqi, 2024-01-26:
The original use case for me is to monitoring the temperature of SSD(internal and external).

exelban, 2024-01-26:
you can propose or PR the sensors mapping