exelban/stats #1805: Fan Speed Irresponsive

Prakhart-debug, 2024-02-05:
**Describe the bug**
The fan speed is not being changed by moving the slider of setting it to max speed as well. Ive reinstalled the app, but the bug still exists. It was working fine a week ago.
Ive attached a screen recording in which the problem is visible.

https://github.com/exelban/stats/assets/75558721/bc70b6e8-41ae-494c-b24d-c37fb13a48dc

**Details:**
 - Device: Macbook Pro 2020 Intel
 - macOS: 14.1.1
 - Application version: 2.9.22

Gea97, 2024-02-05:
If the temperature is too low Fans are disabled by the system

The problem remains also under heavy load?

closed 2024-02-11 by exelban

coldenate, 2024-02-11:
@exelban, is there a specific reason why this has stopped working? It appears that these issues are being closed right away. I'm curious if there was a change made by Apple that is preventing us from fixing it.

exelban, 2024-02-11:
@coldenate yeap, there was an issue with debugging that problem. TL.DR: starting from some macOS update it's not possible to change the fan speed. Looks like Apple blocked that option. I will try to describe all what I know about that in the FAQ next week.

coldenate, 2024-02-12:
@exelban I will say, what I find weird, is that I can get it "unstuck" by reinstalling the Helper. I'll provide a gif when I catch it again. Right now I have it working just fine, but I had to reinstall the Helper.

Prakhart-debug, 2024-02-17:
the 

> @coldenate yeap, there was an issue with debugging that problem. TL.DR: starting from some macOS update it's not possible to change the fan speed. Looks like Apple blocked that option. I will try to describe all what I know about that in the FAQ next week.

hey, so the app Macs fan control works, but the UI of that app sucks. if you can see how they can still control  the fan speed maybe it'll start working again

coldenate, 2024-02-17:
Here's some more info, after I made sure the helper login item was allowed to be enabled on Startup, I haven't had the problem since. It seemed kinda obvious now that I think about it, I wasn't letting the helper tool startup on login, so reinstalling it would simply start the program manually. @exelban 
I'll see if I can get that Gif still now that I have possibly fixed the issue. I'm not able to get it to reproduce though. 🤞 Fingers crossed!