exelban/stats #1807: Stats crashes on every launch after changing settings

chrisheacock, 2024-02-05:
I can't recall which change caused it, but I was tinkering with the display of the info on the status bar and stats crashed. It now crashes on every launch. I have grabbed the latest version and even tried uninstalling and cleaning up, but it still seems to have persistent settings or something.

I do see the information pop onto the status bar for a blip, then it crashes.

Crash dump attached.

[Stats-2024-02-05-131211.ips.zip](https://github.com/exelban/stats/files/14169767/Stats-2024-02-05-131211.ips.zip)

 - Device: MacBook Pro M1 2021
 - macOS: 14.3
 - Application version: 2.9.22

exelban, 2024-02-05:
hi. You can reset the settings by `defaults delete eu.exelban.Stats`.

chrisheacock, 2024-02-05:
So simple!! Thank you and this allowed me to enjoy stats again. This would be a handy thing to throw in the docs just in case someone screws it up like I did.

Thanks agian!

closed 2024-02-05 by chrisheacock