exelban/stats #1815: Tempurature Sensor icon not showing up correctly

JonathanG321, 2024-02-10:
[image]

On the far left the temperature sensor icon will not show up. The other sensor icons work fine.

Clicking on the false icon does bring up the correct dropdown menu as far as I can tell
[image]

JonathanG321, 2024-02-10:
Never Mind this is user Error

closed 2024-02-10 by JonathanG321

daidalo-s, 2024-06-26:
What do you mean by user error? How did you solve it? I'm facing the same problem atm