exelban/stats #1817: [Is this a Bug or Expected Behaviour?] Stats using a lot more of Disk Usage after last update

Gea97, 2024-02-14:
**Describe the bug**
After upgranding to Stats v2.10.0 I noticed that the application has a lot of more Disk Usage, in the last 12 hours it wrote 9 GB on the SSD.

[image]

Maybe this is normal with the new way of reader mode in the latest update

**Details:**
 - Device: MacBook Pro M3 Pro 12/18
 - macOS: Sonoma 14.3.1
 - Application version: 2.10.0

exelban, 2024-02-14:
hi. Yes, it's normal. Stats is using leveldb to save the data starting from the last release.

exelban, 2024-02-14:
do you have enabled sensors module?

exelban, 2024-02-14:
but even with the sensors module it's too much. I need to limit writes to the db. Because right now every read writes to the db.

Gea97, 2024-02-15:
> do you have enabled sensors module?

Yes I do, this is my configuration:

[image]

The update intervals of the modules are the following:

CPU, GPU, RAM, Sensors: 2 sec
Disk: 300 sec