exelban/stats #1820: Crashing upon launch on Sonoma

benbaptist, 2024-02-19:
**Describe the bug**
Stats crashes upon launch, with no warning or message. Only thing that happens is a crash report that shows up in Console.

This previously worked with this exact setup, and no update (on the OS or this app) occured when it stopped working correctly. No settings were adjusted recently. Just stopped launching, seemingly out of the blue. I was on an earlier version of Stats, so I tried updating to the latest `2.10.1` just to be safe, to no avail. 

Here's what I've tried: 

- Rebooting the system does not help. 
- Tried removing `eu.exelban.Stats.plist` from the `Preferences`  folder 
- Removed `Library/Caches/eu.exelban.Stats/`

**Details:**
 - Device: MacBook Air M2 13" (2022)
 - macOS: 14.0
 - Application version: 2.10.1 
 
[log.txt](https://github.com/exelban/stats/files/14326559/log.txt)

exelban, 2024-02-21:
Hi. Thanks for the feedback.
I can make a debug build to detect what exactly causing the problem. Would you like to help?

benbaptist, 2024-02-27:
My apologies for the slow response. If it isn't too much trouble, that would be amazing! Thank you.

closed 2024-03-05 by exelban