exelban/stats #1822: Constantly downloading `Stats.dmg`

tsndr, 2024-02-19:
**Describe the bug**
I've installed `Stats` through `homebrew` but almost everytime I reboot my machine `Stats` won't start on boot (even though it should) and instead puts `Stats.dmg` file into my `~/Downloads` folder.

I don't like this behavior and would like to suggest, that this shouldn't happen when `Stats` is installed through `homebrew`.

**Details:**
 - Device: 2023 MacBook Pro 16"
 - macOS: Sonoma 14.3.1 (23D60)
 - Application version: Can't verify sinc it doesn't start until it is updated.

gurple, 2024-03-03:
Are you using a non-admin account for your daily use or when you experience this? Stats running within a non-admin account's environment typically won't have rights to copy the contents of the fresh .dmg file it downloads. I doubt it has anything to do with Homebrew per se. I also doubt it wise to alter Stats to behave or be packaged differently for Homebrew.

The problem is the design of the internal update logic. It strongly assumes that all environments are admins. It should be checking this state are the very beginning and taking an alternative path of interaction with the user. It really shouldn't just silently download the new DMG and quit without user action.

I've suggested in the past more common approaches used by other software packages.

tsndr, 2024-03-03:
Yes, this occurs on my work computer, which is managed and by my organization and uses only user privileges by default unless I enable a spicific (and timed) admin mode.

exelban, 2024-03-11:
hi. Sorry, but I'm not familiar with the homebrew installation/upgrade process. And does not support it in case of any problems.

nicoluca, 2024-04-09:
Same problem here, and via 'normal' dmg installation and also on a non-admin account, so I agree with @gurple's observation it does not have anything to do with homebrew, and likely rather with the internal update logic?