exelban/stats #1823: Regularly closes/crashes

lukepistrol, 2024-02-19:
For a few days now I noticed that the app regularly quit itself without any crash notification.

After some investigation I found crash reports in `Console.app`:

[image]

I attached the latest report here:

[Stats-Crash-Report.txt](https://github.com/exelban/stats/files/14331739/Stats-Crash-Report.txt)

**Details:**
 - Device: Mac mini M1 16GB
 - macOS: macOS 14.3.1
 - Application version: 2.10.1

mattme, 2024-02-20:
I have the same problem. The relevant lines from the crash log seem to be the following:

```
Process:               Stats [21545]
Path:                  /Applications/Stats.app/Contents/MacOS/Stats
Identifier:            eu.exelban.Stats
Version:               2.10.1 (539)
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2024-02-20 00:00:26.5773 +0100
OS Version:            macOS 14.3.1 (23D60)
Report Version:        12
Anonymous UUID:        DAEB0A66-2DD9-17B2-C6C4-1497B5832E1E

Sleep/Wake UUID:       CAFAB7DC-EB89-4753-8415-0325F8E4A5B9

Time Awake Since Boot: 160000 seconds
Time Since Wake:       45 seconds

System Integrity Protection: enabled

Crashed Thread:        7  Dispatch queue: eu.exelban.Stats.Repeater

Exception Type:        EXC_BREAKPOINT (SIGTRAP)
Exception Codes:       0x0000000000000001, 0x000000019bc9dda4

Termination Reason:    Namespace SIGNAL, Code 5 Trace/BPT trap: 5
Terminating Process:   exc handler [21545]
```

**Details (the same as OP):**
* Device: Mac mini M1 16GB
* macOS: macOS 14.3.1
* Application version: 2.10.1

fLyRhyTHeMz, 2024-03-02:
I have found the same issue occurring for me after years of stable running, last couple of weeks i have been greeted by an empty menu bar requiring a relaunch. 

System is M1 Pro MBP 14.

[image]

> Process:               Stats [66230]
>Path:                  /Applications/Stats.app/Contents/MacOS/Stats
>Identifier:            eu.exelban.Stats
>Version:               2.10.2 (540)
>Code Type:             ARM-64 (Native)
>Parent Process:        launchd [1]
>User ID:               501

>Date/Time:             2024-03-02 03:57:22.2741 -0700
>OS Version:            macOS 14.3.1 (23D60)
>Report Version:        12
>Anonymous UUID:        XXXXXXXXX

>Sleep/Wake UUID:       XXXXXXXXXX

>Time Awake Since Boot: 550000 seconds
>Time Since Wake:       200027 seconds

>System Integrity Protection: enabled

>Crashed Thread:        15  Dispatch queue: eu.exelban.Stats.Repeater

>Exception Type:        EXC_BREAKPOINT (SIGTRAP)
>Exception Codes:       0x0000000000000001, 0x0000000195d49da4

>Termination Reason:    Namespace SIGNAL, Code 5 Trace/BPT trap: 5
>Terminating Process:   exc handler [66230]

exelban, 2024-03-02:
@fLyRhyTHeMz please share the crash report

fLyRhyTHeMz, 2024-03-02:
The whole file? Is that secure?

fLyRhyTHeMz, 2024-03-04:
Crash Report attached.
[Stats-2024-03-04-140014.txt](https://github.com/exelban/stats/files/14487185/Stats-2024-03-04-140014.txt)

benbaptist, 2024-03-04:
I'm having a similar issue, but in my case, it won't even launch. I reported it on #1820, but it might not be the same underlying issue.

exelban, 2024-03-05:
@benbaptist yeap, it looks like you have the same issue.
The crash is caused by the disk module.
I'm trying to catch that crash, but no luck at this moment.

For now there is a build with a disabled Disk module that must prevent crash on start:
[Stats.dmg.zip](https://github.com/exelban/stats/files/14492350/Stats.dmg.zip)

Please let me know if that build crashes or not.

benbaptist, 2024-03-05:
Just tried the disabled disk module version, and indeed, it fires up perfectly! 👍

exelban, 2024-03-05:
thx. I will need some help with debug builds to define where is the problem.
hope that someone can help with that)
I will share the debug build in a few days

lukepistrol, 2024-03-05:
happy to help! I just wanted to let you know that I disabled the Disk module yesterday and it's been running stable since.

exelban, 2024-03-05:
thanks.

a few questions:
- how many drives do you have?
- if the app crashes right on start or after some time?

fLyRhyTHeMz, 2024-03-05:
Indeed this build has been running smoothly since installation this am. 

I normally only have 1 physical drive/ 2 partitions connected via a dock.

lukepistrol, 2024-03-06:
> how many drives do you have?

- 1 internal SSD (1 Partition) - APFS
- 1 external HDD (1 Partition) - APFS
- 1 external SSD (2 Partitions) - APFS
- Multiple SMB shares

> if the app crashes right on start or after some time?

Usually after some time, seemingly at random. When restarting `Stats` it'll work for a couple of hours until it closes/crashes again.

---

However after disabling the Disk module I'm still up and running after almost 2 days now.

exelban, 2024-03-06:
hi. Please try this build:

[Stats.dmg.zip](https://github.com/exelban/stats/files/14513483/Stats.dmg.zip)

exelban, 2024-03-06:
please check this one:
[Stats.dmg.zip](https://github.com/exelban/stats/files/14514401/Stats.dmg.zip)

benbaptist, 2024-03-06:
Perfect! That one works without fail, drives and everything mounted. 👍

[image]

Thank you! Are there any logs you need me to pull for this version to help further any debugging?

exelban, 2024-03-06:
thx. Nope for now, I wanted to see what causing the problem. Will provide a build with logs to test.

exelban, 2024-03-06:
check please this one:
[Stats.dmg.zip](https://github.com/exelban/stats/files/14514869/Stats.dmg.zip)

fLyRhyTHeMz, 2024-03-08:
> check please this one:
> 
> [Stats.dmg.zip](https://github.com/exelban/stats/files/14514869/Stats.dmg.zip)
> 
> 

Can confirm, have been running this build for 2 days now. No crashes with disk module enabled. 

Thanks for the quick fix.

Chamiu, 2024-03-10:
> > check please this one:
> > [Stats.dmg.zip](https://github.com/exelban/stats/files/14514869/Stats.dmg.zip)

no crash.

> how many drives do you have?
1 internal SSD (2 Volume) - APFS, Macintosh HD
1 external HDD (1 Volume) - APFS, Time Machine target
2 external SD (1 Volume) - APFS
1 SMB shares

closed 2024-03-10 by exelban

Chamiu, 2024-05-27:
This issue once more at 2.10.15.
[Stats-2024-05-28-064732.ips.zip](https://github.com/exelban/stats/files/15462009/Stats-2024-05-28-064732.ips.zip)