exelban/stats #1850: Crashing overnight

jadnhm, 2024-03-13:
**Describe the bug**
Stats seems to crash on me overnight regularly

**Details:**
 - Device: MacBook Pro 14 Inch 2021

| Field | Value | 
|------|-------|
|  Model Name |	MacBook Pro |
|  Model Identifier |	MacBookPro18,3 |
|  Model Number |	MKGR3LL/A |
|  Chip |	Apple M1 Pro |

 - macOS: `macOS 14.3.1 (23D60)`
 - Application version: `2.10.4`

[stats_crashes.zip](https://github.com/exelban/stats/files/14587833/stats_crashes.zip)

jpyle490, 2024-03-15:
Same situation on this device:
```
  Model Name:	Mac mini
  Model Identifier:	Mac14,12
  Model Number:	Z170000FYLL/A
  Chip:	Apple M2 Pro
  macOS 14.4 (23E214)
  app version 2.10.4
```
For me it just disappears from the menu bar and I typically don't notice right away. No error or crash dialogs, just no stats app until I restart it.

exelban, 2024-03-17:
hi, sorry but crash reports do not help to understand what caused the crash.

jadnhm, 2024-03-18:
Just out of curiously what /would/ help identify what caused the crash? Happy to try to help pin it down if I can.

exelban, 2024-03-18:
hmm
try disabling modules one by one till the app will not crash. It helps to detect what module causing the crash.