exelban/stats #1862: [Feature Request] add pref to disable color change in network widget

guppy-jpf, 2024-03-25:
having "colorize value" and the ability to choose colors for upload and download in the network widget is very nice.

however, i find it visually distracting when, as the values occasionally go to 0, the pictogram and text become white.
(btw, i'm running in dark mode, with a dark desktop picture, so the menubar text is white by default)

so i'd like to suggest an additional pref, "disable value color change" or something to that effect, such that, when 
enabled, the up/down colors would remain the same regardless of the current value.

thanks for your consideration!

exelban, 2024-04-06:
Hi. Yeap, I think it could be added. Thanks for the feedback.

closed 2024-08-18 by exelban

guppy-jpf, 2024-08-18:
thanks for the implementation!