exelban/stats #1865: Stats showing wrong RAM usage

warmup72, 2024-04-01:
On my MBP M3/macOS 14.4.1 the RAM usage is very different from the activity log.
The free RAM is never the same and difference up to 70%

[image]

closed 2024-04-06 by exelban

warmup72, 2024-04-06:
I find it a bit strange?!?
What is not planned? That stats shows correct values?
I am really very grateful to you for the development, but with the statement "it is not planned for Stats to display correct values" your tool is useless.

Maybe you should explain what your RAM indicator represents?!?

exelban, 2024-04-06:
Maybe you should use a search?

PS: I hate a gh for "not planned", because there is a button: close an issue because of duplicated. But it shows as not planned. All questions to the GH with that.

warmup72, 2024-04-06:
unfortunally I didn't find an open Issue concuring wrong RAM-Display. There are only 2 closed

warmup72, 2024-04-07:
I had a look at other MenuBar tools and realised that all alternatives show the same memory usage as stats. 
The sum of app, reserved and compressed memory is always displayed.
However, this addition is not the same in the Activity Monitor. There is always a more or less large gap to the sum of the occupied memory, which no other tool shows.
Unfortunately, I have not found a description of this gap anywhere

exelban, 2024-04-07:
The reason it's in the Activity Monitor itself. Sum all the values in the Activity Monitor that are included in total usage, and compare it with total usage. And you will see where there is a gap.