exelban/stats #1868: [Feature Request] Add the option to display the real-time network outlet IP on the menu bar

muddlelife, 2024-04-02:
I want to display the export IP of my own network in real time, display it on the menu bar, and increase its frequency, for example, once every 10 seconds. Can you add this option, similar network speed can be displayed on the menu bar? Thank you very much for your contributions to this project

Yuhuadi, 2024-04-02:
But IP addresses are very long... Are you sure you want show it in menu bar? Are you sure it can be show in menu bar?

exelban, 2024-04-06:
Hi.
I think that could be added to the Stats as a new widget that allows visualizing of some parameters: IP address or network name (#1864). But I cannot give any info on when it will be added.

closed 2024-09-15 by exelban