exelban/stats #1874: Incorrect disk space free

jakeelwes, 2024-04-05:
Disk spaced is weirdly very off - showing 300gb (30%) left when disk utility tells me I only have 40gb left (4%).

Thanks for making this app though!!

Macbook M3 Pro 2023
macOS 14.1
stats 2.10.6

closed 2024-04-06 by exelban

warmup72, 2024-04-06:
Maybe same thing like
https://github.com/exelban/stats/issues/1865

and also not planed to show correct value?!?

vnln, 2025-03-26:
@exelban I'm encountering the same issue after updating macos to 15.3.2.

<details>
  <summary>Screenshot</summary>

[image]

</details>

<details>
  <summary>diskutil list output</summary>

[image]

</details>