exelban/stats #1875: Network widget colors issue

mbbatukan, 2024-04-05:
The network arrows' endpoints are not displaying color. Please refer to the attached image.

[image]

**Details:**
 - Device: Macbook Pro 2019 i9
 - macOS: 14.4.1
 - Application version: 2.10.6

closed 2024-04-15 by exelban

mbbatukan, 2024-04-15:
It might be related to the screen resolution and scaling. It's not a big deal, but it does bother me sometimes. thanks for your comment

exelban, 2024-04-15:
I have noticed that there is some bug with drawing the arrow. Will try to fix that, and maybe it will fix that bug.

mbbatukan, 2024-04-15:
thanks a lot