exelban/stats #1881: About contributing to the project

AhmedAbouelkher, 2024-04-06:
I've been using stats for more than 2 years now and I love it.

I recently tried running the source code on my machine and encountered several issues while doing this. 

It would be great if you added a section in [README.md](https://github.com/exelban/stats/blob/master/README.md) showing the steps to clone, explore, and run the project.

exelban, 2024-04-07:
Hi. Thanks for the feedback)
Yeap, I think it could be added.

AhmedAbouelkher, 2024-04-19:
> Hi. Thanks for the feedback) Yeap, I think it could be added.

Are there any updates regarding the build and run process?

exelban, 2024-04-19:
hmm, there must be no problem with building and running. Clone the project -> open Xcode -> select profile -> build/run

exelban, 2024-04-21:
Oh, why do you change the signing certificate?

Try this: target-> build settings->Allow non-modular includes in Framework Modules switch to YES

AhmedAbouelkher, 2024-04-22:
@exelban 
> Oh, why do you change the signing certificate?
> 
> Try this: target-> build settings->Allow non-modular includes in Framework Modules switch to YES

I appreciate your suggestion.

When running the project directly without changing the signing cert to _Sign to Run Localy_ .

<details>
<summary>Screenshot 1</summary>

[image]
</details>

---

I tried changing `Allow non-modular includes in Framework Modules` to `YES` but the issue persists.

> I had to change the signing certificate to _Sign to Run Locally_ for all modules and started the build process.

<details>
<summary>Screenshots 2-3</summary>

[image]
[image]
</details>

exelban, 2024-04-22:
You have selected the signing certificate as the developer id. So you need to specify the team.
Just select `Sign to Run Locally` and team as `None`.

AhmedAbouelkher, 2024-04-22:
> You have selected the signing certificate as the developer id. So you need to specify the team. Just select `Sign to Run Locally` and team as `None`.

Already did that and tried to build the project, but then I encountered errors in [Screenshots 2-3](https://github.com/exelban/stats/issues/1881#issuecomment-2068720859).

exelban, 2024-06-30:
Hi.
I just have a fresh install on a new Mac. And everything works just fine without any tweaks. So I don't see a reason for that.

closed 2024-06-30 by exelban