exelban/stats #1883: [Feature Request] Memory Pressure Indicator

DeltaScratchpad, 2024-04-08:
It would be beneficial if the RAM indicator could optionally report the memory pressure, in the same way as allocation.

For myself I'd like it to replace the current count/percentage value, but others may like to have both values displayed.

Current allocation based counting may suggest that the system is under more load than it actually is, due to RAM being used for caching etc. Memory pressure would represent a more accurate account of system utilisation.

Currently the expanded panel shows a rough value, but one which more closely matches the Activity Monitor display would be perfect.

Thank you for your work on this great tool!

exelban, 2024-04-15:
Hi, it will be possible after #1868

closed 2024-04-15 by exelban

ShevonKuan, 2024-05-11:
> Hi, it will be possible after #1868
The indicator can only show the pressure by changing its color. Is it possible to show the percentage of the pressure? Like istatistica pro which uses `Wired + Used` as the memory pressure.