exelban/stats #1885: Can not show the battery level of bluetooth keyboard

jack24254029, 2024-04-10:
**Describe the bug**
Can not see the battery level of bluetooth keyboard

[image]
[image]
[image]

**Details:**
 - Device: Mac mini 2023
 - macOS: 14.4.1
 - Application version: 2.10.7

igorsantos07, 2024-04-14:
The same happens with my mouses. As a test, I connected a Logitech MX Master 3 (which doesn't even show up on the widget, nor is disabled in the Stats settings) and a generic one from AliExpress:

[image] [image]

Mine is an Intel Core i5-8500B, same versions as the OP, and using the standard bluetooth chip.

exelban, 2024-04-15:
Hi. I have to take a look at that. And there is not so much I can do. Looks like Apple broke the best API that gives BLE percentage. Looks like for now there is no good API for ble.
There always was a problem with getting a BLE percentage. And now I'm thinking about dropping this module support. Because there is no good way to get this information(

closed 2024-04-15 by exelban

jack24254029, 2024-04-17:
I think can only show the battery level about apple product for now, ex: Magic Trackpad, Magic Keyboard
Other ble devices can wait.