exelban/stats #1889: [Feature Request] Improvements for Notifications and Swap usage

jvanakker, 2024-04-11:
Some requests:

- For example swap usage: we can currently set a warning notification in Stats. However, this notification has no hysteresis or limit check. Once the threshold is reached, Stats continuously keeps pushing new notifications. There seems to be no check if it already created a notification a second ago. Meaning there will be hundreds of notifications so you currently need to disable them manually every time in Stats.

- Stats (2.10.7) lets me set a warning at 0.5 GB of swap, but can this be changed so we get a notification with ANY swap usage above zero?

- The possibility to show the actual swap value in the widget would be great. Currently you have to click on RAM widget, then see the swap there.

Swap is relatively important for users with SSD's (in other words, any modern mac), because they have a limited lifespan (number of writes), and when macOS is using swap, it has a significant effect on the (soldered) SSD lifespan.

M1 macOS 14.3.1

exelban, 2024-04-11:
Hi. Thanks for your feedback.
1 - Looks like bad behavior of the notifications, needs to check that.
2 - Yeap, I can add an option > 0
3 - in theory it could be added after #1868 is closed. As result there will be a new widget that allows to show anything you want from the module in the menu bar.

closed 2024-06-24 by exelban

bakjoul, 2025-03-02:
> Hi. Thanks for your feedback. 1 - Looks like bad behavior of the notifications, needs to check that. 2 - Yeap, I can add an option > 0 3 - in theory it could be added after [...]

Hello. Can we show used swap now in the menu bar ? I'm looking for this feature but I can't seem to find the swap usage option for the module.