exelban/stats #1897: Swap data read (blue) and data written (red) in the Disk module

checktext00, 2024-04-16:
In the Disk popup the data read (blue) is on the bottom and the data written (red) is on top. I think the default should be the other way around (like how it's done in Activity Monitor's Disk section - written data is "downloaded" to the disk and read data is "uploaded" from the disk)

Anyway - I tried to change it but I noticed the only option to change it in the Widget section (so I get what I want in the actual macOS menu bar) but there's no option to change it in the popup

Also I realize that the colors in the Stats app icon are also the "wrong" way around (according to my definition). Maybe it could be changed? But the issue is really the lack of the option for the popup

[image]
[image]

Anyway, thanks for working on this app and the quick responses for issues

checktext00, 2024-04-16:
I also realized I never see any processes in my "Top processes" in the Disk popup - is that normal?

exelban, 2024-04-16:
why my it's bad, and activity monitor is good?
You can change the colors if you don't like defaults.

checktext00, 2024-04-17:
> why my it's bad, and activity monitor is good? You can change the colors if you don't like defaults.

I just meant good and bad as an example of what I wrote (bad = the issue I wanted to fix)

I don't want to change the colors, I want to change the order for written data to be on the bottom and read data to be on top in the popup (the same way it is in the menu bar icon in my previous post - so basically just to add "Reverse order" to the Disk Popup menu)

[image]
[image]

exelban, 2024-04-17:
oh, yes. This option could be added.

checktext00, 2024-04-18:
> oh, yes. This option could be added.

ok thanks

exelban, 2024-04-18:
yes

exelban, 2024-04-18:
yes

closed 2024-04-28 by exelban

checktext00, 2024-05-01:
fixed, thanks