exelban/stats #1900: Download / Upload graphs incorrectly swapped in Network Popup

briankelly, 2024-04-17:
**Describe the bug**
Download traffic is displaying as upload traffic. See screenshot below, where I'm downloading 1-2 MB/s and the graph is displaying that download traffic as upload traffic (in red).

I reset my Stats settings to default and restarted the app before reproducing this. It happens consistently. 

[image]

**Details:**
 - Device: Macbook Pro (M3 Pro)
 - macOS: Sonoma 14.4.1
 - Application version: 2.10.9 ([release](https://github.com/exelban/stats/commit/30dd07116e5ed4e4373d265013843f448e944944))

exelban, 2024-04-17:
hi. Thx for your feedback. Will be fixed.

SirLyoshyk, 2024-04-20:
@exelban The same issue with the disk read/write activity. Could it be fixed too or should I open a new issue?

closed 2024-04-21 by exelban

briankelly, 2024-04-22:
Confirmed the fix landed in [v2.10.10](https://github.com/exelban/stats/releases/tag/v2.10.10). Thank you!