exelban/stats #1902: Network menubar graph drawing incorrectly

alext, 2024-04-18:
**Describe the bug**

Hi, since upgrading to 2.10.9 (from 2.10.7), I'm seeing the network graph in the menubar rendering oddly:

[image]

2 things not right here:
* The up and down graphs are using different scaling making it appear that they're equal when they're not.
* The graphs aren't filled in as they used to be. 

Intererstingly, if I swap the order of the network panels, the graph is filled in again:

[image]

**Details:**
 - Device: Macbook Pro 2021 - M1 Max
 - macOS: Ventura 13.6.6
 - Application version: 2.10.9

alext, 2024-04-18:
> And what about the chart in the popup?

The rendering looks fine in there. They have the same scaling behaviour though (each being independently scaled).

I seem to recall there was an option in the previous version to control whether they used a common scale or not, but that looks to have gone with this version...

thanks for looking into this...

exelban, 2024-04-18:
could you try to reinstall the app?

alext, 2024-04-18:
> could you try to reinstall the app?

That doesn't seem to have made any difference (I `brew uninstall`ed and then reinstalled with `brew install`).

exelban, 2024-04-18:
hmm, looks like some bug in the oldest version of macOS
I will try to debug that.

exelban, 2024-04-21:
Hi, I have checked stats on macOS 13 and do not notice that bug.
Also looks like there are no users with similar bug.

closed 2024-04-21 by exelban

Kvantti, 2024-09-29:
I'm having the same issue as @alext - the widget network graphs are not scaled uniformly:
[image]

- MacBook Pro Apple M3 Max 14-inch Nov 2023
- macOS Sonoma 14.7 (23H124)
- Application version 2.11.8

The popup graph has an option to set Main chart scaling as "Fixed" and set a user setting for Scale value, which allows to correct the graph scaling for that view. 

Would it be possible to include the option for fixed scaling and scale value for the widget graphs as well? That seems to be an easy fix for the users experiencing this bug.