exelban/stats #1904: Since a few builds, the option of the network views to have the same scale for upload and download is gone

bouscram, 2024-04-19:
Everything is in the title... 

As an addition, an option to set the scale to a fixed value (for instance the interface bandwidth) would be appreciated...

closed 2024-04-20 by exelban

bouscram, 2024-04-21:
Hi @exelban, did you close the issue from the title or the additional option request? (sorry, I shouldn't have put two things in the same ticket)

exelban, 2024-04-21:
take a look at faq first

bouscram, 2024-04-21:
OK, I see that there the fixed network scale is in the works. 
But I didn't find anything regarding the fact that the "same scale for up and down" option has disappeared. I am sure it was here very clearly. Is this disappearance linked to the preliminary steps for the fixed scale?

exelban, 2024-04-21:
that one (common scale) is not available for now.