exelban/stats #1908: Combined modules not displaying

theangrydev, 2024-04-22:
[image]
[Stats.plist.zip](https://github.com/exelban/stats/files/15055919/Stats.plist.zip)

eric-saintetienne, 2024-04-27:
You're lucky to get all of them because even when not combining them I get only a couple of them displayed, somtimes the batter, sometimes the network, sometimes the cpu or the bluetooth, but never all of them like you.

exelban, 2024-05-02:
Hi. Hmm, strange. I have imported your settings and everything works fine. Need to debug that. If you can reproduce that problem at any time I can make a debug build that will tell more what is going on.

theangrydev, 2024-05-03:
@exelban sure happy to retry with a debug build, this reproduces consistently for me

exelban, 2024-05-05:
hi. Thanks for your help.
There is a debug build:
[Stats.dmg.zip](https://github.com/exelban/stats/files/15213035/Stats.dmg.zip)

Please run this build. It will collect some logs to the `Documents/log.txt`. Please share these logs afterward.

closed 2024-05-17 by exelban