exelban/stats #1910: Version update bug

selenawoodward, 2024-04-25:
**Describe the bug**
A clear and concise description of what the bug is.
Starting from 2.10.4, pushed updates will not change the version number
**Details:**
 - Device: [e.g. Macbook Ari M1 2020]
 - macOS: [e.g. 13.6.5]
 - Application version: [e.g. 2.10.4]
[image]

exelban, 2024-04-25:
there is no such problem.

closed 2024-04-25 by exelban

selenawoodward, 2024-04-25:
https://github.com/exelban/stats/assets/99124972/95d57b9b-4c5c-4940-9488-6a9583ef4169

I did find this bug on my MacBook indeed. Here's the video recording of the bug. I'm using the Chinese language, but after the update, you can see that the version number hasn't changed

selenawoodward, 2024-04-25:
No worries at all! It happens sometimes. It's great that you found the new version in your downloads folder. If you need any further assistance or have any more questions, feel free to ask.

selenawoodward, 2024-04-25:
I'm sorry for wasting your time, I found a new version of the .dmg in the download folder, my question, I thought it would update automatically

exelban, 2024-04-25:
it must, but depending on your Mac config it could or it could not.