exelban/stats #1915: Uptime mismatch

andyw8, 2024-04-29:
**Describe the bug**

My terminal shows:
```
12:08  up 13 days, 15:18, 16 users, load averages: 6.46 6.70 6.18
```

But Stats shows:

[image]

**Details:**
 - Device: Apple M1 Pro
 - macOS: Sonoma 14.4
 - Application version: 2.10.11

exelban, 2024-04-29:
please share the screen of the system report with the uptime

andyw8, 2024-04-30:
Not sure if you mean this screen? Or something else?

[image]

exelban, 2024-04-30:
https://iboysoft.com/wiki/mac-uptime.html

andyw8, 2024-04-30:
Yeah, it matches what's shown in the System Information. I don't understand why, but it seems this isn't a bug in Stats so I'll close.

closed 2024-04-30 by andyw8