exelban/stats #1923: Shows Nothing for Sensors Widget

MhmRhm, 2024-05-01:
On a fresh Mac and fresh install, I  go to Sensors and enable it and also enable any subsystem like fans or temperature. Nothing shows up in the widget.

The 2.10.11 has this issue but 2.10.10 is fine.

**Details:**
 - Device: MacBook Pro M2 Pro 14"
 - macOS: Sonoma 14.4.1
 - Application version: 2.10.11
 
[image]

exelban, 2024-05-01:
Hi. Thanks for your feedback.
Yeap, it's a bug. Will be fixed.

closed 2024-05-05 by exelban

DennisHaijma, 2024-06-06:
This is still broken on my M1 Max I think, it definitely worked before. I see the issue was resolved/completed. I am on the last update (2.10.11) but still see similiar display of sensors output as in the provided screenshot of @MhmRhm.

DennisHaijma, 2024-06-06:
This is still broken on my M1 Max @ Ventura 13.0. It definitely worked before. I see howevever that the issue was resolved/completed. I am on the last update of Stats (2.10.15) but still see similiar display of sensors output as in the provided screenshot of @MhmRhm. Reset of settings did not help.