exelban/stats #1928: Disk Widget does not show data from correct volume

PuzzledUser, 2024-05-03:
I have four auto-mounted volumes. Downloading a file onto my startup disk shows the activity in two of the other volumes. My startup volume shows zero traffic. See attached.
Stats Version 2.10.11
3.5 GHz Quad-Core Intel Core i5
Retina 5K, 27-inch, Late 2014 
Fusion drive
Mac OS 14.4.1 (23E224)
OpenCore 1.4.3

My startup volume is not the first volume in **diskutil list**:

```
% diskutil list
/dev/disk0 (internal, physical):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:      GUID_partition_scheme                        *121.3 GB   disk0
   1:                        EFI EFI                     209.7 MB   disk0s1
   2:                 Apple_APFS Container disk2         121.1 GB   disk0s2

/dev/disk1 (internal, physical):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:      GUID_partition_scheme                        *3.0 TB     disk1
   1:                        EFI EFI                     209.7 MB   disk1s1
   2:                 Apple_APFS Container disk2         3.0 TB     disk1s2

/dev/disk2 (synthesized):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:      APFS Container Scheme -                      +3.1 TB     disk2
                                 Physical Stores disk0s2, disk1s2
   1:                APFS Volume Macintosh HD - Data     1.2 TB     disk2s1
   2:                APFS Volume Preboot                 3.5 GB     disk2s2
   3:                APFS Volume Recovery                2.4 GB     disk2s3
   4:                APFS Volume VM                      1.1 MB     disk2s6
   5:                APFS Volume Another                 184.0 GB   disk2s7
   6:                APFS Volume Other HD                84.2 GB    disk2s9
   7:                APFS Volume Macintosh HD            15.3 GB    disk2s10
   8:                APFS Volume Startup - Data          794.2 GB   disk2s11
   9:                APFS Volume Startup                 12.2 GB    disk2s12
  10:              APFS Snapshot com.apple.bless.2D31... 12.2 GB    disk2s12s1

```

[image]

PuzzledUser, 2024-05-07:
No change in 2.10.12. (retested, just FYI)

exelban, 2024-05-19:
hi, I will try to fix it. But since it's only a problem on hackintosh it will be not in a prior.

PuzzledUser, 2024-05-20:
Hi, thanks. My assumption was that it's due to booting from disk2s11 rather than the more usual disk2s1. I updated the screenshot above to make the names clearer

closed 2024-08-12 by exelban