exelban/stats #1932: Disk widget Update Interval menu doesn't change update interval

PuzzledUser, 2024-05-07:
The disk module has an update interval option. However, updates remain at the default of 1 sec irrespective of changes made. 

A similar control in the CPU and RAM modules does alter the update interval.

**Details:**
Stats Version 2.10.12
3.5 GHz Quad-Core Intel Core i5
Retina 5K, 27-inch, Late 2014
Fusion drive
Mac OS 14.4.1 (23E224)
OpenCore 1.4.3

closed 2024-05-07 by exelban

PuzzledUser, 2024-05-07:
Haha, I guess it works for others...

PuzzledUser, 2024-05-12:
Despite https://github.com/exelban/stats/issues/1942 fixed in https://github.com/exelban/stats/releases/tag/v2.10.13, the Disk update interval does not change at all for me.

exelban, 2024-05-12:
What do you mean?

PuzzledUser, 2024-05-12:
You have done work on another bug relating to disk update interval not persisting. It seemed perhaps a related effect that for me, the update interval remains at default irrespective of changes to the setting.

exelban, 2024-05-12:
How do you debug that?

PuzzledUser, 2024-05-12:
Haha. Er, I see the updates continue to happen every 1 second even when I change the update Interval control to some other value. I have throughput displayed — so it's really obvious:
[image]

https://github.com/exelban/stats/assets/91260935/81a151db-e5bd-4380-8174-b56e44a2b168

PS. Note the same fast 1 sec interval on the Network request [I also posted about](https://github.com/exelban/stats/issues/1933) that you didn't fancy implementing.

exelban, 2024-05-12:
yes, everything works fine. Update interval it's about the capacity. Not activity. KB/s means per second. It means reading every second.

PuzzledUser, 2024-05-12:
OK, we'll have to agree to differ. To me, it's perfectly reasonable to want to see the activity per second, but averaged over the chosen interval, so it is easier to see the trend (rather than the noise). 

This is how Mac OS Activity Monitor works — see View -> Update Frequency.