exelban/stats #1941: two issues of disk IO: update interval not working, system processes not showing

Dzhami-Jame-ALI, 2024-05-11:
**Describe the bug**
There are probably two issues in disk IO widget: 

1. The `update interval` settings does not work. Whatever value I choose, the menu wieget will always update every 1 second. 
2. The `Top processes` 's disk IO is not showing macOS system process like `mds_stores`. The I/O from top processes is significant different than menu bar icon's value in that case, e.g. top bar shows 60MB/100MB R/W (actually by `mds_stores`), but top processes shows nothing. 

**Details:**
 - Device: macbook m3 max 16''
 - macOS: 14.4.1
 - Application version: 2.10.12

exelban, 2024-05-12:
Duplicate of #1942

closed 2024-05-12 by exelban

Dzhami-Jame-ALI, 2024-05-13:
@exelban Hello, for #issue2, it might be related to root permission: 
```
            let result = withUnsafeMutablePointer(to: &usage) {
                $0.withMemoryRebound(to: (rusage_info_t?.self), capacity: 1) {
                    proc_pid_rusage(pid, RUSAGE_INFO_CURRENT, $0)
                }
            }
            guard result != -1 else { return }
```
If root permission is granted when building the program, those system processes like `mds_stores` will get reulst `0` rather than `-1`. 
Is it possible to consider adding a root-based helper as an option to this tool, like fan-helper?

exelban, 2024-05-13:
First, if the guard check does not pass it means that Stats cannot get usage info.
Second, nope. I'm avoiding the root access. And for the feature like that I will not add it for sure.

Why are you running the app under the root?

Dzhami-Jame-ALI, 2024-05-13:
No offense, I'm just trying to get the reason why iStat Menus can get those system processes' disk IO (probably it requires installing a root-based helper). Sometimes when seeing the huge disk IO numbers in menus bar, it's natural to want to click and see the details.

exelban, 2024-05-13:
No worry, I'm ready to help with fixing that. Looks like you run the Stats from the source?

exelban, 2024-05-13:
I have a few reports that users do not have processes for disk. But have no idea why and do not see that problem on any of my Macs.

Dzhami-Jame-ALI, 2024-05-13:
Yes, I changed the scheme to debug the process as root. 
If debugged as root, then system processes will appear after filter `guard result != -1 else { return }` .