exelban/stats #1942: Update interval setting is not saved/restored on application restart

hemanth981, 2024-05-11:
**Describe the bug**
If any of the update interval is changed, the changes hold good only until the application/OS is restarted. For example, if we change CPU/sensor widget to update every 2 seconds instead of 1 sec, it works fine until you restart the app or reboot your Mac. After restarting, it resets to the default value of 1 sec. I checked and this happens to the update interval of all widgets.

I'm not sure if this is relevant, but I installed Stats from brew.

**Details:**
 - macOS: 14.4.1
 - Application version: 2.10.12

exelban, 2024-05-11:
hi, thanks for the report.
Yeap, its a bug, will fix that.

closed 2024-05-12 by exelban

hemanth981, 2024-05-12:
Wow that was fast! Thanks for the excellent app!